HuskieRobotics/roborioExpansion #14: Add log resume ability

wiresboy, 2015-03-30:
# DO THIS NOW
Basically make the command that saves and closes the file also clear the buffers and prepare to do a new log file. This way, it creates a new file each time we enable.
Reseting the SD card cog may be easiest, IDK exactly what will work best.

lmrezac, 2015-10-21:
The function will be function number 0x04.

lmrezac, 2015-10-21:
Alright, I think I did it. If command 0x04 is received, the value of the sd log data will be set to "stop". Once that happens, the sd card logger will notice it and cease executing it's mainLoop, calling it's "reinit" method, wiping the logger cog's stack, stopping the cog, and starting a new cog . IT will also wipe the file name value, so this command should be followed by the function 0x03 (set file name).

closed 2015-10-21 by lmrezac