HuskieRobotics/roborioExpansion #40: LCD Newline characters

wiresboy, 2017-01-04:
To go to the beginning of the next line, the Parallax LCD expects the bytes `\r\n` (0x13 0x10). However, a new line on linux systems (like the roboRIO) only has the `\n` part, which means that data written to the LCD isn't formatted as expected. The cursor will move to the next line, but will not reset to the beginning of the line. 

Possible solution: Automatically replace the default newline character with an explicit `\r\n`.

@gcschmit Do you think we should allow this to be turned off with a boolean input? The default would obviously be to replace the character.

gcschmit, 2017-01-04:
What value is there to move to the next line and not reset to the beginning of the line? I could see value in resetting to the beginning of the line without moving to the next line, perhaps to update a value. I think we should always replace for now. If we get requests for this feature, it can be added later.

Would the replacement occur in the LV API or in the firmware?

wiresboy, 2017-01-05:
This change happens in LabView. The reason I suggest this, is because it is pretty easy to write at a specific location (Row 2, Col 4, etc.). In the past, we have written the "static" stuff out once (units, names, etc.) then updated only the values that we needed to. Otherwise, there is a pretty terrible flicker if you clear the entire screen every time.

I'd say make the replacement, then we can close this issue and add the option to disable this later.

gcschmit, 2017-01-05:
Makes sense. We should add a feature for an advanced LCD VI that allows the user to address a specific starting row and column. Or, add optional terminals to the current VI for a specific starting row and column.

wiresboy, 2017-01-05:
Sure, that is another thing we can do. In the short term, if they buy the Parallax LCD, they will be given directions with the LCD on what control bytes do what, and they can embed those in their own labview code.

gcschmit, 2017-01-05:
Modified HB_DisplayLCD to replace newlines with DOS-style newlines.

Commit: 1536f107b0fc8a744bdaaa9fc7c6a35fc4a61a2f

closed 2017-01-05 by gcschmit