HuskieRobotics/roborioExpansion #41: Make LCD clear-screen optional

wiresboy, 2017-01-04:
Currently we force the LCD screen to clear on every packet. While this makes sense if you don't know what data had been there before, this is fairly restrictive to advanced users. 

I suggest that we move the clear-screen byte to the labview end (away from the propeller), and then have an enable/disable boolean input for the vi.

As this could be a major incompatibility issue if we don't fix this before flashing the boards, I am going to mark this as a high priority.

wiresboy, 2017-01-05:
Completed Huskie Board FW fix. 78cf00fa6da24c8acd3fff4165ed4325da2455b5

wiresboy, 2017-01-05:
@Hobbs2000: the change to the LabView VI is: When the user wants the screen to be cleared, prepend a "0x00" byte, otherwise transmit the data without any extra byte. The default should be to transmit the clear byte.

gcschmit, 2017-01-05:
@Hobbs2000 I'm writing the LCD Display example right now. I have to make some changes to HB_DisplayLCD anyway; I'll add this as well.

gcschmit, 2017-01-05:
I updated HB_DisplayLCD to add an optional input for this feature. Leaving open since documentation is incomplete.

Commit: 1536f107b0fc8a744bdaaa9fc7c6a35fc4a61a2f

gcschmit, 2017-01-07:
The control bytes are documented in the HB_DisplayLCD VI.

@lmrezac This must also be documented in the LabVIEW API User Guide and then this can be closed.