HuskieRobotics/roborioExpansion #46: Prevent GPIO from being an output on certain pins.

wiresboy, 2017-01-04:
We do not want to accidentally directly control certain pins, because they could interrupt normal configuration. This will be some simple checks in labview and on the propeller. This should only affect setting a pin as an output, reading inputs from all pins is still ok.

Do not allow outputs on pins:
 - [ ] 0-6: SD card
 - [ ] 19-23: ADC
 - [ ] 28-31: EEProm, USB - serial

wiresboy, 2017-01-04:
The labview implementation should have a typedef that only includes the allowed pins.

6. roboRIO MOSI
7. roboRIO MISO
8. roboRIO CLK
9. roboRIO CS
12. roboRIO SCL
13. roboRIO SDA
14. GPIO 0
15. GPIO 1
16. GPIO 2
17. GPIO 3
18. LCD
24. LED 0 (Green)
25. LED 1 (Green)
26. LED 2 (Green)
27. LED 3 (Red)

gcschmit, 2017-01-04:
Which pins map to the DIP switches?

wiresboy, 2017-01-04:
Fixed.

Pin | Name
---|------------
6 | roboRIO MOSI / DIP Switch 4
7 | roboRIO MISO / DIP Switch 3
8 | roboRIO CLK / DIP Switch 2
9 | roboRIO CS / DIP Switch 1

gcschmit, 2017-01-04:
Is the overlap between the SPI pins and the DIP switches okay because we don't use SPI to talk to the roboRIO?

AssemblyCode, 2017-01-04:
The board has silkscreen D1, D2, D3, D4 for the LEDs and the switch itself
is correspondingly marked 1, 2, 3, 4.  Please update your names accordingly.

wiresboy, 2017-01-04:
@gcschmit Correct.

@AssemblyCode: I grabbed those numbers from the code's pinmap, where they appear to be working. Maybe there is a weird reverse happening in the code somewhere, I will investigate.

AssemblyCode, 2017-01-05:
Thanks.  Could be left over from rev1 board, or was arbitrarily assigned.

gcschmit, 2017-01-05:
Changing to high-priority since a new typedef changes the LV API. It would not be a backward compatible change.

wiresboy, 2017-01-05:
Well, right now there is no typedef. Don't they just output an integer? Wouldn't we have it map its outputs 1:1 with the pins?

wiresboy, 2017-01-05:
Implemented in Spin, so removing that tag.
9b7bbc350e172f3ff55a82ec61a25dd32d95bfcc

gcschmit, 2017-01-05:
Technically, the type def is a different type than an integer. I know you can wire an integer into an input that expects a typedef and it will be coerced. I'm not sure if you change the input of a VI from an integer to a typedef if it will break existing code. I'll investigate in a moment as I'm working on the LV examples tonight as Calvin made many of the changes already to the API VIs.

gcschmit, 2017-01-05:
Are names for all 31 pins defined somewhere? I want the typedef to reflect all 31 pins with those we don't want disabled....

gcschmit, 2017-01-05:
I created a typedef for all the GPIO pins. However, it needs to be updated with the appropriate labels. In addition, HB_ConfigureDIO needs to be updated to generate an error if the specified pins are set for output.

wiresboy, 2017-01-05:
PIN | Description | Alternate | allow output
--- | --- | --- | ---
0 | sd_d1 | | no
1 | sd_d0 | | no
2 | sd_clk | | no
3 | sd_cmd | | no
4 | sd_d2 | | no
5 | sd_d3 | | no
6 | RR MOSI | switch 1 | Yes
7 | RR MISO | switch 2 | Yes
8 | RR CLK | switch 2 | Yes
9 | RR CS | switch 1 | Yes
10 | RR RX | | no
11 | RR TX | | no
12 | RR SCL |  | Yes
13 | RR SDA |  | Yes
14 | GPIO 0 |  | Yes
15 | GPIO 1 |  | Yes
16 | GPIO 2 |  | Yes
17 | GPIO 3 |  | Yes
18 | LCD |  | Yes
19 | adc_CS2 | | no
20 | adc_CS1 | | no
21 | adc_DO | | no
22 | adc_CLK | | no
23 | adc_DI | | no
24 | LED 0 | (Green) | Yes
25 | LED 1 | (Green) | Yes
26 | LED 2 | (Green) | Yes
27 | LED 3 | (Red) | Yes
28 | EEPROM SCL | | no
29 | EEPROM SDA | | no
30 | USB TX | | no
31 | USB RX | | no

As it turns out, the switches were labeled in the reverse order in all of the SPIN code. That has now been fixed, along with the places that relied on those switches.

AssemblyCode, 2017-01-05:
Please update names for LED 1,2,3,4 and Switch 1,2,3,4, matching them
physically on the board. We want to be assembly-centric and ignore any
schematic names.

wiresboy, 2017-01-05:
As I said, the mistakes in the code are fixed, the names are now the same in the code, schematic, assembly, and documentation.

AssemblyCode, 2017-01-05:
But I see this above.  It's been updated?
...
6 RR MOSI switch 1 Yes
7 RR MISO switch 2 Yes
8 RR CLK switch 2 Yes
9 RR CS switch 1 Yes
...
24 LED 0 (Green) Yes
25 LED 1 (Green) Yes
26 LED 2 (Green) Yes
27 LED 3 (Red) Yes
...

closed 2017-01-05 by wiresboy