HuskieRobotics/roborioExpansion #51: File Resource Contention

gcschmit, 2017-01-05:
What is the behavior when a user doesn't work with files in the way we intend. For example, each of the following (at least) actions need to have a defined response:

* writes to a file without opening it
* closes a file without opening it
* opens a file when one is already opened

While we only support a single file opened at a time, we may still want to use the refnum model to help us manage this resource and leave the door open for future expandability. A refnum would provide the information necessary to respond to each of these cases.

Thoughts?

AssemblyCode, 2017-01-05:
> @AssemblyCode <https://github.com/AssemblyCode>.

I see assignments to me (@AssemblyCode).  What is your intention.  I don't
know LV or Spin.

gcschmit, 2017-01-05:
@AssemblyCode just want your thoughts on behavior.

wiresboy, 2017-01-05:
The HuskieBoard keeps track of if a file is open, and doesn't allow itself to do anything "bad" on the SD card, like writing data w/out having a file open.

1. Writes to a file, but no file is open: Nothing is written, nothing crashes.
2. Closes a file without opening it: Nothing closes, nothing crashes.
3. Opens a file while another is open: Closes the first, then opens the new one.
4. Opens a file with a filename that is invalid: Depends on how invalid. If it is too long, it crops the filename to the right length. The right set of characters could cause a file to be inaccessible via windows (It is visible, but not readable/deletable). This should be limited by the LabView file open command.

wiresboy, 2017-01-05:
Also, I don't think we will ever allow multiple files to be open. Currently we are using some open source code to deal with the filesystem on the Huskie Board, and it only supports one file at a time. Allowing multiple would be a large project in and of itself. 

I don't think we need to make any functional changes, so I am closing this issue. If the purpose of this issue was to remember to clarify this in the documentation, then please reopen.

gcschmit, 2017-01-05:
Reopening so I can verify the LV API does the right thing and the documentation captures this behavior.

gcschmit, 2017-01-07:
This behavior is documented in the HB_OpenLog, HB_CloseLog, and HB_WriteToLog VIs.

@lmrezac This must also be documented in the LabVIEW API User Guide and then this can be closed.