HuskieRobotics/roborioExpansion #54: Replace error values in LV API with standard NI error values

gcschmit, 2017-01-05:
If we use our own custom error values, they need to be in the user-defined range and we have to document them so they are compatible with the Explain Error VI. For now, I think we should find the closest existing NI error value to use instead.

wiresboy, 2017-01-05:
I find it really useful to get errors like "Incorrect Checksum" or "Timeout waiting for response". If there are no exact LabView matches, then I would strongly suggest that we keep ours. 

How hard is it for us to define our own range? We have to document the meaning of the errors when the come from us vs LabView anyway.

gcschmit, 2017-01-05:
We have to create a HB-errors.txt file and install it in the proper location. I worry about the installation part. Let me see what standard LV errors I can find. We only have two that I've noticed so far: bad checksum and string too long.

Here are the VIs that need to be revisited. I may have missed some:

Need to update error codes to standard LV errors:
	HB_ConfigureDIO (bad checksum)
	HB_DisplayLCD (bad checksum)
	HB_ReadDIO (bad checksum)
	HB_ReadAISingleChannel (bad checksum)
	HB_ReadAIAllChannels (bad checksum)
	HB_writeToLog (string was too long)

gcschmit, 2017-01-05:
http://zone.ni.com/reference/en-XX/help/371361L-01/lvhowto/def_custom_error_text/

gcschmit, 2017-01-07:
I created a custom error file for the Huskie Board. It currently contains two values:

5000: Bad checksum received from Huskie Board. Retry operation.
5001: String length too long. Check the documentation from the generating VI for the maximum length of the string that is supported.

This file needs to be installed in user.lib/errors/ The errors directory may need to be created.

@lmrezac These must be documented in the LabVIEW API User Guide.

This issue should only be closed after the documentation is completed and the installer supports installing this file.