HuskieRobotics/roborioExpansion #55: LED strip setup

wiresboy, 2017-01-05:
Add a packet to the API to:
 - Select the LED output pin (With constraints - only the pins listed below)
 - LED Type ("WS2812" or "WS2812B", since the timings for both are slightly different). 
 - Length of strip (Up to 256)

Allowable pins:
 - SPI pins
 - I2C pins
 - GPIO pins
 - LCD pin

gcschmit, 2017-01-05:
Would this packet also include the RGB values? If not, how will we associate RGB values with the right pin?

Basically, I'm wondering if we need to create a HB_ConfigureLED VI and a HB_SetLEDRBGValue VI or if we just need the later. If the former, what will be used to connect the two calls? The pin?

wiresboy, 2017-01-05:
We will need to have two calls. We don't need to have anything differentiate them, since we can only support 1 strip at a time.

wiresboy, 2017-01-05:
More of a limitation of time. The current library we are using only supports 1 strip. There are other libraries that support more, or we could use more cogs, but both require some fairly significant changes to how the current driver is used. We can investigate a change when we get to implementing "modes" or something along those lines.

gcschmit, 2017-01-05:
I created a LED Model and LED Pins typedefs. I also created HB_ConfigureLED, HB_SetLEDIntensity, and HB_SetLEDRGBValue VIs. The HB_ConfigureLED VI doesn't compile since the serial api isn't defined for that functionality yet.

wiresboy, 2017-01-06:
Ok, HB_ConfigureLED has now been implemented on both the roboRIO (commit 797d3893ab6364b343a1bd7e3c4fba1b80615885) and Huskie Board (commit e8c81f382b8a53184b3ee2e53493e66cae07c553).

It is also documented in the Software Spec

closed 2017-01-06 by wiresboy