HuskieRobotics/roborioExpansion #60: Update DIO Pins typedef and LED Pins typedef with standard names

gcschmit, 2017-01-05:
@wiresboy If you can provide a table of pins and names, I will update the typedef.

wiresboy, 2017-01-05:
PIN | Description | Alternate | allow output
--- | --- | --- | ---
0 | sd_d1 | | no
1 | sd_d0 | | no
2 | sd_clk | | no
3 | sd_cmd | | no
4 | sd_d2 | | no
5 | sd_d3 | | no
6 | RR MOSI | switch 1 | Yes
7 | RR MISO | switch 2 | Yes
8 | RR CLK | switch 2 | Yes
9 | RR CS | switch 1 | Yes
10 | RR RX | | no
11 | RR TX | | no
12 | RR SCL |  | Yes
13 | RR SDA |  | Yes
14 | GPIO 0 |  | Yes
15 | GPIO 1 |  | Yes
16 | GPIO 2 |  | Yes
17 | GPIO 3 |  | Yes
18 | LCD |  | Yes
19 | adc_CS2 | | no
20 | adc_CS1 | | no
21 | adc_DO | | no
22 | adc_CLK | | no
23 | adc_DI | | no
24 | LED 0 | (Green) | Yes
25 | LED 1 | (Green) | Yes
26 | LED 2 | (Green) | Yes
27 | LED 3 | (Red) | Yes
28 | EEPROM SCL | | no
29 | EEPROM SDA | | no
30 | USB TX | | no
31 | USB RX | | no

As it turns out, the switches were labeled in the reverse order in all of the SPIN code. That has now been fixed, along with the places that relied on those switches.

AssemblyCode, 2017-01-05:
LED still shows 0,1,2,3 and needs to be 1,2,3,4 to match switches.

wiresboy, 2017-01-05:
Thank you. Fixed.

wiresboy, 2017-01-05:
Typedef is now accurate.

closed 2017-01-05 by wiresboy