HuskieRobotics/roborioExpansion #65: Create set time VI

wiresboy, 2017-01-06:
It was implemented in our old code, but isn't around any more. It shouldn't be too hard to implement.

I'm thinking that we should use the current roboRIO time, and just make sure to document that this VI shouldn't be called too early, since the roboRIO doesn't have a RTC onboard and instead relies on the driverstation to get the time.

We need to also document that this VI must be called **before** the file is opened. Otherwise it won't change the "last modified" tag on the file.

gcschmit, 2017-01-06:
If we are going to use the current roboRIO time and not let the user specify the time, which is fine, why don't we incorporate this behavior into the HB_OpenLog VI? Then we don't have to worry about sequencing issues.

gcschmit, 2017-01-06:
Regardless, I did create a HB_SetTime VI. I modeled it after other VIs, but I am not confident that I packed the bits correctly. Specifically, I don't appreciate why the array of booleans is being reversed. Please review.

wiresboy, 2017-01-06:
I like the HB_open_log idea, we would need to make it optional though. If the rr hasn't connected to the dashboard, then I don't think that the time is defined.

wiresboy, 2017-01-07:
The bytes, after reversing, were individually in the wrong order. For now, I added another set of reverses, but we should consider this issue still open.

The HB code works though, fortunately.

gcschmit, 2017-01-07:
I updated HB_OpenLog to automatically invoke the HB_SetTime VI by default.

@wiresboy Would you re-run the Data Logging Example to make sure it still works?